lvgl/lv_port_stm32f469_disco #2: The project cannot be compiled as is, LVGL used is very old

MR-DOS, 2020-06-22:
Hi,
the project cannot be compiled as is - one has to modify line 41 in "stm32f4xx_it.c" from
`#include "lvgl/lv_hal/lv_hal_tick.h"`
 to
`#include "lvgl/src/lv_hal/lv_hal_tick.h"`

Also, line 45 in "main.c" has to be commented out (the function called doesn't exist) while line 44 has to be uncommented.

The project is compatible only with version of LVGL up to v5.3 while current version is v7.0.2. Therefore the usability of this project for starting with LVGL is limited - one has to rewrite it to support the current versions.

kisvegabor, 2020-06-23:
Hi,

Were you able to make it work? If you can you send a  Pull request?

MR-DOS, 2020-06-24:
Hi,
believe it or not, but I don't know how to send a pull request. I'm not a programmer, but a HW engineer who suffers every time when having to deal with programming. 
All that is needed to make the project bompilable is incuded in my initial report of the issue - only minor changes on 3 lines are needed.

kisvegabor, 2020-06-25:
> believe it or not, but I don't know how to send a pull request.

No problem :slightly_smiling_face: 
Are you interested in learning it? We have a short guide about it [here](https://docs.lvgl.io/latest/en/html/contributing/index.html#pull-request).

MR-DOS, 2020-06-29:
OK, so I tried replacing the old 6.x LVGL with 7.x LVGL. All it takes is checking out the current version along with the examples, correcting one broken link to the "lv_conf_checker.h" of LVGL to "lv_conf_internal.h" which does the same, it was just renamed. Then replacing the old "lv_conf.h" and "lv_ex_conf.h" with the new templates, uncommenting their content and setting the resolution to 800x480.
It is also possible to very easily enable DMA2D, but id didn't result in any speed-up in my case...

kisvegabor, 2020-06-30:
Thanks!

Can you attach you `lv_conf.h`? Just to be sure I don't miss any settings.

closed 2020-08-01 by stale[bot]