lvgl/lv_port_stm32f469_disco #3: Unable to integrate LVGL port for STM32F469i DISCO into an STM32 Project

praman-matter, 2021-03-31:
I tried using the port as suggested in :
https://blog.lvgl.io/2017-07-15/stm32f429_disco_port

But it seems to create the port as a separate directory/project with its own main.c. 

So how to use this and verify graphics? The path is not visible to my STM32F's main.c. I cannot invoke init for tft from my main.c.. 

The main.c in the port remains unexecutabe - so it seems the make file also needs to be changed? 

Could you please guide me ?

Thanks!
Pavitra

kisvegabor, 2021-03-31:
Hi, 

This project wasn't updated for a while however it should work as it is. Have you used SWSTM32 (Ac6) IDE to open it?

praman-matter, 2021-03-31:
Yes I opened the project under ATM32 ac6 tool chain (STM Cube IDE) ..
Should I send you the whole project to look at?

The main project just toggles LEDs on board (LED1,2,3 and 4)

Regards
Pavitra

praman-matter, 2021-03-31:
These are the kinds of errors-arm-none-eabi-gcc "../src/stm32f4xx_it.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DSTM32 -DSTM32F4 -DSTM32F469NIHx -DDEBUG -DSTM32F469xx -DUSE_HAL_DRIVER -c -I"/Volumes/GoogleDrive/My Drive/STM/Projects/LVGL_test/lv_port_stm32f469_disco_sw4stm32-master" -I"/Volumes/GoogleDrive/My Drive/STM/Projects/LVGL_test/lv_port_stm32f469_disco_sw4stm32-master/inc" -I"/Volumes/GoogleDrive/My Drive/STM/Projects/LVGL_test/lv_port_stm32f469_disco_sw4stm32-master/Utilities/STM32F469I-Discovery" -I"/Volumes/GoogleDrive/My Drive/STM/Projects/LVGL_test/lv_port_stm32f469_disco_sw4stm32-master/HAL_Driver/Inc/Legacy" -I"/Volumes/GoogleDrive/My Drive/STM/Projects/LVGL_test/lv_port_stm32f469_disco_sw4stm32-master/HAL_Driver/Inc" -I"/Volumes/GoogleDrive/My Drive/STM/Projects/LVGL_test/lv_port_stm32f469_disco_sw4stm32-master/CMSIS/core" -I"/Volumes/GoogleDrive/My Drive/STM/Projects/LVGL_test/lv_port_stm32f469_disco_sw4stm32-master/CMSIS/device" -I"/Volumes/GoogleDrive/My Drive/STM/Projects/LVGL_test/lv_port_stm32f469_disco_sw4stm32-master" -O3 -ffunction-sections -Wall -fstack-usage -MMD -MP -MF"src/stm32f4xx_it.d" -MT"src/stm32f4xx_it.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "src/stm32f4xx_it.o"
../src/stm32f4xx_it.c:41:10: fatal error: lvgl/lv_hal/lv_hal_tick.h: No such file or directory
   41 | #include "lvgl/lv_hal/lv_hal_tick.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [src/subdir.mk:30: src/stm32f4xx_it.o] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

embeddedt, 2021-03-31:
Have you also cloned/updated submodules?

praman-matter, 2021-03-31:
I just have what is there on the github repo

Regards
Pavitra

embeddedt, 2021-03-31:
How did you download it from the GitHub repo?

praman-matter, 2021-03-31:
I downloaded it from
https://github.com/lvgl/lv_port_stm32f469_disco_sw4stm32

Regards
Pavitra

praman-matter, 2021-03-31:
Pavitra Ramanujam
2:38 PM (0 minutes ago)
to *lvgl/lv_port_stm32f469_disco_sw4stm32*
I downloaded it from
https://github.com/lvgl/lv_port_stm32f469_disco_sw4stm32

Downloaded as a ZIP flle and expanded it and copied into an STM32F AC6
project's root folder

Regards

embeddedt, 2021-03-31:
Ah; I see. Downloading as a ZIP won't work as we use submodules; you'll need to use `git clone --recursive`.

embeddedt, 2021-04-01:
You should be able to clone recursively and then just import the project by itself. It shouldn't be within your own project.

praman-matter, 2021-04-01:
Thanks, I'll check it out. I was not able to open a separate project. I
will try some more and let you know

praman-matter, 2021-04-01:
Hi

I performed the following actions -

At the terminal of my workspace root folder :  git clone --recursive
https://github.com/lvgl/lv_port_stm32f469_disco_sw4stm32.git

Then I tried to import the project lv_port_stm32f469_disco_sw4stm32 into my
cubeIDE workspace as AC6 project. But nothing happened

praman-matter, 2021-04-01:
Hi
Is the project not STMCube IDE? Is it something else? Is that why it does
not open ?

Please clarify

Regards
Pavitra

embeddedt, 2021-04-01:
It wasn't migrated to CubeIDE, but it is a proper SW4STM32 (the previous version) project so it should be able to be imported. :confused:

praman-matter, 2021-04-01:
Well It is not being recognized as a project. I have tried this. I will try
again. But f not, any suggestions what can be done? How to assemble from
scratch for instance?

Thanks
Pavitra

closed 2021-06-22 by stale[bot]