lvgl/lv_port_stm32f469_disco #5: Stm32Cube IDE project

avdept, 2021-09-16:
Hey folks. Any chance to get stm32cube ide project file(ioc) to make it easier to do config? I tried to migrate code from this project into fresh project, but end up with having really unstable project(UI laggs and sometimes freezes completely) even though I copied all settings I could find in code and all files.

kisvegabor, 2021-09-16:
Hi,

Can you share that project. It could be a good starting point.

avdept, 2021-09-16:
Sure. I'll give it 1 more day, to try to get it up and running. In any case I'll push the project to github and share link here, so skilled folks could have a look on what did I do wrong.

avdept, 2021-09-16:
@kisvegabor Actually I've pushed everything I got here https://github.com/avdept/VentsController

So 2 issues I'm facing
* Overall laggy UI using printer demo. I'm running it on F469 disco board, and running it using project from this repo has no issues.
* Vertical line on right side of screen(I'm not sure if its because of incorrectly set display dimensions somewhere or something else)

avdept, 2021-09-16:
I just checked with another demo(music player) and it seems that part of display being shifted a little, see ex here https://ibb.co/NYcktPn

LuWeiHuan, 2021-10-30:
Hey friend, This modification can solve your [UI lags] problem
[image]

As for the black bar on the left of your screen, I didn't check it for you, but it's certain that there is a problem with your display driver code

closed 2021-11-05 by avdept