lvgl/lv_port_stm32f469_disco #6: Building with CubeIDE 1.11.2 gives errors

sp5wwp, 2023-01-30:
Errors:
```
c:\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.100.202210260954\tools\arm-none-eabi\bin\ld.exe: ./hal_stm_lvgl/tft/tft.o:C:/Users/SP5WWP/Desktop/lv_port_stm32f469_disco_sw4stm32/Debug/../hal_stm_lvgl/tft/tft.c:81: multiple definition of `hltdc_eval'; ./Utilities/STM32F469I-Discovery/stm32469i_discovery_lcd.o:C:/Users/SP5WWP/Desktop/lv_port_stm32f469_disco_sw4stm32/Debug/../Utilities/STM32F469I-Discovery/stm32469i_discovery_lcd.c:135: first defined here
c:\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.100.202210260954\tools\arm-none-eabi\bin\ld.exe: ./hal_stm_lvgl/tft/tft.o:C:/Users/SP5WWP/Desktop/lv_port_stm32f469_disco_sw4stm32/Debug/../hal_stm_lvgl/tft/tft.c:80: multiple definition of `hdsi_eval'; ./Utilities/STM32F469I-Discovery/stm32469i_discovery_lcd.o:C:/Users/SP5WWP/Desktop/lv_port_stm32f469_disco_sw4stm32/Debug/../Utilities/STM32F469I-Discovery/stm32469i_discovery_lcd.c:136: first defined here
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:149: stm32f469_discovery_no_os.elf] Error 1
"make -j16 all" terminated with exit code 2. Build might be incomplete.
```

How to reproduce:
- clone the repo with `git clone --recursive https://github.com/lvgl/lv_port_stm32f469_disco_sw4stm32`
- import the directory as a project with CubeIDE
- agree to convert to a CubeIDE-compatible project
[image]
[image]
- run `Build All`

kisvegabor, 2023-02-01:
I agree that it'd be better to have a CubeIDE project, however, unfortunately I don't have this board now. Could you give a try to create CubeIDE project from a scratch and I can help if you stuck somewhere.

sp5wwp, 2023-02-01:
The problem is that I lack the required skills to do this on my own :) I'd clone the repo and offer a PR already, if I knew how to do it. My assumption is that all the required code is contained by this project already. All we need to do is to move it to CubeIDE-generated one. If you are willing to try, I can run it on my DISCO board.

kisvegabor, 2023-02-04:
I've just tried it but it seems a little bit more complicated. I've also tried to convert the project with CubeIDE but it just doesn't do anything when I click the "Finish" button. 

Could you send your project. Probably it's easier to fix the build error :slightly_smiling_face:

mdiepart, 2023-02-07:
Hi,

I think this might be because the name of the variables in the tft.c compile unit is the same as the name in the stm32469i_discovery_lcd.c compile unit. 
I prefixed the name of the variable with an ad-hoc prefix however I do not have the required board to test it.

mdiepart, 2023-02-08:
He already tried to compile the project but got a black screen. 
I should receive a board to test for that in the coming days and will keep you posted.

We will try to create a native CubeIDE project too.

mdiepart, 2023-02-13:
It seems that the fix should work however I found that the touch sensor does not initialize properly and crashes the mcu.
Has this code been successfully executed in the past?

I am currently porting lvgl 8.3 in a CubeMX project.

kisvegabor, 2023-02-14:
Note that `lv_indev_drv_t` and `lv_diso_drv_t` should be `static` in v8.

Maxoos, 2023-02-21:
@mdiepart did you get it sorted on CubiIDE? also don't have the screen working but it compiled ok.

mdiepart, 2023-02-21:
I managed to get a CubeMX project to work on the board however I am not happy with the way the DSI and LTDC peripherals are handled (I had to use the BSP otherwise it wouldn't work).

I had to wait for PR lvgl/lvgl#4004 to be merged.

The result is in repo https://github.com/mdiepart/lv_port_stm32f469_disco_cubemx, I do not know how we should proceed from here...

Maxoos, 2023-02-21:
> I managed to get a CubeMX project to work on the board however I am not happy with the way the DSI and LTDC peripherals are handled (I had to use the BSP otherwise it wouldn't work).
>
> I had to wait [...]

Thanks, it's working. Did you see that strange diagonal line?

kisvegabor, 2023-02-23:
I agree that ideally we should be able to use the display without the BSP library,. however it's already a great step as it allows the users to open the project in the official IDE.

So if you sent a PR I'd be happy to merge it. :slightly_smiling_face:

kisvegabor, 2023-02-23:
#7 is merged so I close this issue. 

Please open a new issue if you find any problem.