lvgl/lv_port_stm32f469_disco #8: Build problem on STM32F469_dicso

nicho98, 2023-03-27:
Hi, I'm currently working on th STM32F469 discovery board and I'm trying to implement LVGL on it in STM32CubeIDE. However, after dowloading the source code of LVGL for STM32F469 and importing the LVGL library from the master(version 9.0.0), I was unable to build the project due to some errors. 
Can anyone help me understand what I'm doing wrong?
Thanks in advance for your time and feedback.
[image]
- Build Error
[image]
- Disable the DMA 2D in lv_conf.hLV_USE_GPU_STM32_DMA2D 0
[image]

TechStranger, 2023-03-28:
Hi Gabor. Thank you for your prompt response. I imported the 8.3ver of LVGL and it builds fine (lv_demo_benchmark();). However, when I tried to run the project, I got a black screen. I'm not sure if this is the problem with my code as i tried it with touchGFX's project, it works perfectly fine.

kisvegabor, 2023-03-29:
What happens if you run is in Debug mode. Does it halt somewhere?

TechStranger, 2023-03-29:
When i run the project in debug mode, it halts at HAL_Init(), indicating No source available for "__udivmoddi4() at 0x80005a4" 
[image]

kisvegabor, 2023-03-31:
I cc @mdiepart  who added a large update to this project recently. 

@mdiepart  could you take a look at this issue?

mdiepart, 2023-04-02:
Hi,
First of all, would you mind checking and giving us your versions of STM32CubeIDE, the STM32F4 FW as well as the version of your board?

Regarding what you see when running in debug mode, this is not a bug. By default the code stops at the first line of the main function which is `HAL_Init();` So what you see is the code being paused (as indicated in the left panel of the screen: 
`Thread #1 [main] 1 [core: 0] (Suspended: Breakpoint)`).

mdiepart, 2023-04-02:
Also, you should not manually import the lvgl library, it is included as a submodule and you have to retrieve them using these commands:
```sh
git submodule init
git submodule update
```

This will ensure that you have the adequate library version at the required location.

TechStranger, 2023-04-03:
Hi @mdiepart, I am currently using version 1.12.0 of STM32CubeIDE and have attempted to obtain the lvgl library by following the instructions you provided.
> git clone https://github.com/lvgl/lv_port_stm32f469_disco.git
> cd lv_port_stm32f469_disco
> git submodule init
> git submodule update

After executing the program on the STM32 target, the screen remains black and nothing is displayed.

[image]

mdiepart, 2023-04-07:
Hi,

For the FW version, can you please look at the version you use and tell us? Latest version is 1.27.1.

For the board version it is written on the board and should look like DK32F469I$AU1. What you showed us is the version of the st-link chip which is used to upload firmware to the board.

TechStranger, 2023-04-11:
Hi @mdiepart ,
Sorry for the late reply,
I verified the FW for F4, it's running on 1.27.1 and about the board version, the sticker on the rear side of the board is indicating "MB1189 **B-01**", i think it's the older version of the Discorvery Board. 
Will the board's version affect how LVGL works?

[image]
Thanks

mdiepart, 2023-04-11:
This might indeed be a problem. This example was written with the rev C. Can you try to compile the example by doing the following modification?

Right click on the project -> Properties -> C/C++ General -> Paths and Symbols -> "Symbols" tab. On the right click "Add". Name, set to `USE_STM32469I_DISCO_REVB`. Value: leave empty. Tick "Add to all configurations".

This should compile the code with support for the display controller used on the revision B of the board which is different from the display controller of rev C.

Keep us posted of how it turns out :)

TechStranger, 2023-04-11:
Hi @mdiepart. I did the modification and it works. Thanks for your help

closed 2023-04-12 by TechStranger